icub-tech-iit/robot-snapshots #12: ergocub_sn_000 - Modify CAD Snapshot

GiacomoVassallo13, 2022-11-16:
**Applicant:** *Giacomo Vassallo, @GiacomoVassallo13 

**Snapshot name:** *ergocub_sn_000*

**Field of application:**
- [X] Mechanics
- [ ] Electronics
- [ ] Wiring

## Request Update Description
Modify the CAD Snapshot of ergoCub, added a washer in the yaw joint of the neck. See the comments : 

- https://github.com/icub-tech-iit/proto/issues/2001#issuecomment-1313344124
- https://github.com/robotology/icub-tech-support/issues/1233#issuecomment-970462707

[Optional table]

| Item / IITCODE | Quantity |  Description | Notes  |
|:---:|:---:|:---|:---|
| 6287 | 1pz | NECK SPACER (int. diameter 26 mm) |  |
| 1874 | 4pz | Washer - flat regular type - M2.5 (2.7x6x0.5) - SS A2 140HV - DIN125 A, ISO7089, UNI 6592 |  |
|  |  |  |  | 
|  |  |  |  |

## Definition of Done
[insert here the definition of done for this task]

GiacomoVassallo13, 2022-11-16:
Hi @Fabrizio69. I wrote 5 in the confusion of corrections. There are 5 washers in total, one above the bearing (the bigger one) and the other four to compress it against it (the small ones), as shown in the figure. The countersunk screws remain free because they hold the sensor support, the other two do not have the washer as this would make them collide against the support.

cc: @Elio75

Fabrizio69, 2022-11-17:
> These two screws are not installed? [image]
>

Hi @fiorisi these two screws are used for the sensor support (missing in the photo).

Elio75, 2022-11-18:
- I created a 6287 part as a neck assembly ring (assembly IC_014_G_181 new rev A2 ), 

[image]
[image]

-  You have to understand why this compenetration exists in CAD

[image]

From the CAD you can see the bearing and the two Kaydon bearings fit well in the seats.

[image]

but the Kaydon bearing still overhangs by 0.1 mm as you can see from the picture below

[image]

Talking to @GiacomoVassallo13  he told me that this interpenetration doesn't appear, otherwise the AEA bearing might stick out

@maggia80 @Fabrizio69 @fiorisi  @GiacomoVassallo13

fiorisi, 2022-11-18:
If the CAD is correct the spacer shouldn't be used. Is in contact with both the outer and inner ring.
[image]

I think that we have to design a proper spacer and add it to the official CAD model, not only to the snapshot.

Elio75, 2022-11-18:
`If the CAD is correct the spacer shouldn't be used. Is in contact with both the outer and inner rings.`

[image]

@maggia80 @Fabrizio69 @fiorisi @GiacomoVassallo13

Fabrizio69, 2022-11-18:
Hi @Elio75 it's better to use a spacer that touch only the outer ring to avoid sliding

cc: @fiorisi

Elio75, 2022-11-18:
But the bearing rests on the inner ring inside its seat

[image]

CC @Fabrizio69

Elio75, 2022-12-14:
chiudo

closed 2022-12-14 by Elio75