icub-tech-iit/robot-snapshots #13: ergocub_sn_000 - lower arms implementation

Lawproto, 2022-11-18:
**Applicant:** *Lorenzo Protopapa*, @Lawproto 
cc @fiorisi 

**Snapshot name:** *ergocub_sn_000*

**Field of application:**
- [x] Mechanics
- [ ] Electronics
- [ ] Wiring

## Request Update Description
Implement both full lower arms - currently missing - in the robot. Namely:
- `ECUB_004_A_004.ASM`
- `ECUB_004_A_005.ASM`

found in `ergocub-design-lowerarm repository`, `mech` branch

[Optional table]

| Item / IITCODE | Quantity |  Description | Notes  |
|:---:|:---:|:---|:---|
|  |  |  |  |
|  |  |  |  |
|  |  |  |  | 
|  |  |  |  |

## Definition of Done
Above mentioned aliases implemented with the right positioning.

Lawproto, 2022-11-18:
@Fabrizio69 @fiorisi in my opinion, pull requests have a good potential for these kind of tasks. They act as an issue while simultaneously prompting the applicant to provide the right material themselves.  Worth investigating.

Fabrizio69, 2022-11-18:
Ciao @fiorisi, can we evaluate @Lawproto 's proposal?

pattacini, 2022-11-18:
An issue is much slimmer and serves to notify everybody that something will start soon.
To open a PR, instead, you need to do already some work.

Therefore, the ideal case is to open an issue beforehand and then do the work in a branch/fork to finally open a PR. [Issues and PR's can be linked to each other](https://docs.github.com/en/issues/tracking-your-work-with-issues/linking-a-pull-request-to-an-issue).

fiorisi, 2022-11-18:
> Ciao @fiorisi, can we evaluate @Lawproto 's proposal?

Discussed F2F with @Lawproto.
The main issue is that the file are binary, and the reviewer would have to change the remote or clone the fork. We will commit the files and keep this issue open to align with @Elio75.

pattacini, 2022-11-18:
>The main issue is that the files are binary,

That's true and binary files cannot be compared within the PR dashboard of course.
However, a PR still represents a moment in time when a delivery is complete, thus a resource that can be referenced and used later.

Just wanted to highlight all the pro's and cons'.

Lawproto, 2022-11-18:
@Elio75 I backupped `ecub_004_a_004` and `ecub_004_a_005` in this repo, ergocub_sn_000 branch, and committed (see above).
They have to be mounted on the robot, waiting for your feedback.

Elio75, 2022-11-23:
Both lower arms fitted

[image]

CC @Fabrizio69 @Lawproto

Lawproto, 2022-11-24:
@Elio75 do you think we can close this one?

closed 2022-11-25 by Elio75