icub-tech-iit/robot-snapshots #20: [ergoCub_sn_000] [introducing shims for LCORE positioning]

Elio75, 2023-01-27:
**Applicant:** *Elio Massa*, @Elio75 

**Snapshot name:** *platform_sn_xxx*

**Field of application:**
- [ ] Mechanics
- [ ] Electronics
- [ ] Wiring

## Request Update Description
[Introduction of shims for placement of LCORE left hip]

[Optional table]

| Item / IITCODE | Quantity |  Description | Notes  |
|:---:|:---:|:---|:---|
|  |  |  |  |
|  |  |  |  |
|  |  |  |  | 
|  |  |  |  |

## Definition of Done
[insert here the definition of done for this task]

Elio75, 2023-01-27:
Two shims were added for the positioning of the Lcore board in order to be able to read the left hip optical disc lower body.

[image]

Work in progress

@Fabrizio69

Elio75, 2023-02-28:
we disassembled the left leg to check the validity of the optical disk, the latter was found to be rifled so it was not a problem with the sensor placement of the Lcore board so the shims were removed.

CC @Fabrizio69

Fabrizio69, 2023-03-03:
@Elio75 we can close this issue

closed 2023-03-03 by Elio75