icub-tech-iit/robot-snapshots #21: [platform_sn_xxx] [request issue title]

Elio75, 2023-02-08:
**Applicant:** *NElio Massa*, @Elio75

**Snapshot batch2020-01** *iCub2.7_sn_051*

**Field of application:**
- [ ] Mechanics
- [ ] Electronics
- [ ] Wiring

## Request Update Description
[Rimuovere dallo snapshot batch2020-01 icub_sn_046 i seguenti componenti]

[Rimuovere]

| Alias / IITCODE | Quantity |  Description | Notes  |
|:---:|:---:|:---|:---|
| RC_IIT_018_G_070 / 5723 |4  | SEA joint relative encoder assembly | Rimuovere dall'assieme RC_IIT_018_G_001 / RC_IIT_018_G_050 / RC_IIT_018_G_031 / RC_IIT_018_G_003  |
| RC_IIT_018_P_050 / 5015 | 4 | Magnetic strip support |  Rimuovere dall'assieme RC_IIT_018_G_016 / RC_IIT_018_G_026 |
| MAGNETIC_STRIP_V3 / 5678 | 4 | Magnetic strip, 2mm pole length | Rimuovere dall'assieme RC_IIT_018_G_016 / RC_IIT_018_G_026 | 
| RC_TLR_003_P_101 / 3492 | 2 | THIMBLE FOR CABLE Ø1.75mm | Rimuovere dall'assieme RC_TLR_001_G_031 |
| V3-10--_-_U5931_GC / 2448 | 1 | screw – M 3 x 10  – UNI5931 , DIN912  , ISO4762 – SS A2 – hexagon socket cap head - fully threaded | Rimuovere dall'assieme RC_IIT_018_G_009, vite superiore sx che tiene la piastra RC_IIT_018_P_020   |
| V3-10--_-_U5931_GC / 2448 | 1 | screw – M 3 x 10  – UNI5931 , DIN912  , ISO4762 – SS A2 – hexagon socket cap head - fully threaded | Rimuovere dall'assieme RC_IIT_018_G_028, vite superiore dx che tiene la piastra RC_IIT_018_P_020   |

## Definition of Done
[insert here the definition of done for this task]

Elio75, 2023-02-08:
Il nuovo branch icub_sn_051 è stato creato a partire da batch_2020-01.

work in progress

CC @Fabrizio69

Fabrizio69, 2023-02-09:
Ciao @Elio75, lo snapshot `icub_sn_051` dovrà contenere:
- Lowerbody, Hands and Forearms del `batch_2020-01` con le modifiche [sopra](https://github.com/icub-tech-iit/robot-snapshots/issues/21#issue-1576248704) indicate
- Upperbody (RC_TLR_003_A_005_UPPERBODY without hands and forearms), Upperbody battery backpack covers (RC_IIT_017_A_011) del [BATCH_2021_A](https://github.com/icub-tech-iit/robot-snapshots/tree/icub_BATCH_2021_A/cad-model)
- Head `TO BE DEFINED`

Fabrizio69, 2023-02-09:
@Elio75 fai attenzione alle cover che forse non sono presenti in tutti gli snapshot che ho citato nel [mio commento](https://github.com/icub-tech-iit/robot-snapshots/issues/21#issuecomment-1423871683) precedente.

Fabrizio69, 2023-02-09:
@Elio75 , riporto qui di seguito i serial number dei vari sotto assiemi:
- Head: To Be Defined
- Upperbody: S/N:048U
- Hands+Forearms: S/N:052HF
- Lowerbody: 043LW-044L

Elio75, 2023-02-10:
## Attenzione

icub_batch2020-01 ha gli FTsens 45 in REV A3
icub_batch2021_A ha gli FTsens 45 in REV A4
Il nuovo snapshot icub_sn_051 che ho creato è un ibrido tra icub_batch2020-01 e icub_batch2021_A.  Unendo tutti file nello stesso branch il sn_051 è rimasto con gli Ftsens 45 in REV A4, questo è dovuto al fatto che i file possedevano lo stesso nome perciò unendo i file si sono sovrascritti quelli della REV A3. 

CC @Fabrizio69

Elio75, 2023-02-13:
## Aggiornamenti

[image]

- Ftsens 45 cod. 12487 REV A4 
- (upperbody) Riduttore Nidec shimpo cod. 14458 e 14459
- Upperbody Back-Pack cover, alias RC_IIT_025_G_100, cod 7455 rev A5.
- Upperbody Cover battery backpack assy, alias RC_IIT_025_G_102, cod. 7157 rev A4

Elio75, 2023-02-15:
## Pezzi modificati 

RC_TLR_011_P_051 cod. 3320.A1

[image]

RC_TLR_012_P_034 cod. 3277.A1

[image]

CC @Fabrizio69

Fabrizio69, 2023-03-31:
Ciao @Elio75 ,  riporterei nel [README.md](https://github.com/icub-tech-iit/robot-snapshots/tree/icub_sn_051/cad-model#readme) l'immagine di cosa contiene lo snapshot: cioè un robot coverizzato ma ancora senza testa e chiuderei questa issue.

Elio75, 2023-04-07:
[README iCub_sn_051](https://github.com/icub-tech-iit/robot-snapshots/edit/icub_sn_051/README.md) aggiornato

CC @Fabrizio69

closed 2023-04-07 by Fabrizio69