icub-tech-iit/robot-snapshots #24: [ergoCub_SN_000] [modifying RealSense flat support and retainer]

Mick3Lozzo, 2023-03-01:
**Applicant:** *Michele Gesino*, @Mick3Lozzo 

**Snapshot name:** *ergoCub_SN_000*

**Field of application:**
- [ ] Mechanics
- [ ] Electronics
- [ ] Wiring

## Request Update Description
Redesigning the components, and adding the pivot point, I am going to re-send them to production. See [reference issue](https://github.com/icub-tech-iit/fix/issues/1326).

[Optional table]

| Item / IITCODE | Quantity |  Description | Notes  |
|:---:|:---:|:---|:---|
| IC_014_P_151 - 15730.A1 | 1 | ergoCub head, RPLiDar support, flat cable RealSense locker |  |
| IC_014_G_197 - 16514.A2 | 1 | ergoCub head, RPLiDAR USB adapter, support |  |

## Definition of Done

- [ ] Replace `IC_014_P_151 - 15730` with` IC_014_P_151 - 15730.A1`
- [ ] Replace IC_014_G_197 - 16514.A1 with `IC_014_G_197 - 16514.A2`

Mick3Lozzo, 2023-03-01:
@Fabrizio69 the WinGST BOM and orders have to be updated, correct?

Fabrizio69, 2023-03-03:
@Mick3Lozzo the WinGST BOM has to be updated. Orders are currently for CUST and SHTM parts only, so no changes are needed for the components in the [table](https://github.com/icub-tech-iit/robot-snapshots/issues/24#issue-1605207902)

Elio75, 2023-03-16:
CAD UPDATE DONE

[image]

cc @Fabrizio69 @Mick3Lozzo

closed 2023-03-20 by AntonioConsilvio