icub-tech-iit/robot-snapshots #30: [ergocub_sn_000] [Montaggio covers gruppo principale alias ECUB_007_A_001 ]

Elio75, 2023-04-21:
**Applicant:** *Elio Massa*, @Elio75 

**Snapshot name:** *ergocub_sn_000*

**Field of application:**
- [ ] Mechanics
- [ ] Electronics
- [ ] Wiring

## Request Update Description
Verrà montato a CAD sullo snashot in data 21/04/2023 le covers gruppo principale di ergocub. Visto che da quando sono state progettate fino ad oggi hanno subito delle modifiche, perciò esiste la possibilità che ci siano delle divertsità tra gruppo covers e la meccanica.

[Optional table]

| Alias / IITCODE | Quantity |  Description | Notes  |
|:---:|:---:|:---|:---|
| ECUB_007_A_001 / 16928 |1  | ergoCub, covers, main assembly |  |
|  |  |  |  |
|  |  |  |  | 
|  |  |  |  |

## Definition of Done
Aggiornamento snapshot ergocub_sn_000

Elio75, 2023-04-21:
1- I due pezzi (staffe) ECUB_007_P_047 servono per fissare la maschera (visore) ECUB_007_P_029 al telaio della testa.
 
[image]

A livello CAD questi fori sono  riferimenti di fissaggio dell'assieme covers della testa alias ECUB_007_G_060

Elio75, 2023-04-21:
(2)/(3)- Questi due supporti della parte inferiore della gamba sia destra come sinistra hanno i fori  di fissaggio come riferimento per il montaggio delle covers anteriori

Elio75, 2023-04-21:
3- Questi due supporti posteriori della parte inferiore della gamba sia destra come sinistra hanno i fori di fissaggio come riferimento per il montaggio delle covers posteriori o sia del pezzo Alias ECUB_007_P_037 cod. 16914.A1

Elio75, 2023-04-26:
L'alias ECUB_007_G_030 sarebbe il supporto della scheda **CONFU_DISPLAY-SOCKET**, questo supporto in teoria era montato sul gruppo covers della testa ECUB_007_G_060 rev 0.

[image]

Questo supporto all'inizio doveva essere fatto di lamiera ma aveva un problema di compenetrazione con il supporto della Realsense. Per ovviare a questo problema il pezzo è stato realizzato in plastica aumentando anche lo spessore ed evitando cosi problemi di montaggio con il display OLED. Essendo fatto in plastica li permetteva di flettere evitando cosi tensioni sul display OLED durante il collegamento tra il connettore e la scheda. 
È stato anche rialzato con dei distanziatori standoff da M3x5 mm di lunghezza) per dare più ricchezza alla flat che collega la scheda CONFU_DISPLAY-SOCKET con la scheda DIGITALVIEW_MIP-1000. [vedere issue](https://github.com/icub-tech-iit/project-ergocub/issues/79) .

 > Essendo fatto in plastica li permetteva di flettere evitando cosi tensioni sul display OLED durante il collegamento tra il connettore e la scheda. 

Su questo problema c'è questa [issue](https://github.com/icub-tech-iit/project-ergocub/issues/103#issuecomment-1513302316) che spiega come potrebbe essere risolto il problema spostando di 4.5 mm i fori di fissaggio della scheda verso il visore

Elio75, 2023-04-27:
Ho montato i pezzi che mancavano per poter piazzare in modo corretto il gruppo cover ECUB_007_A_001 sullo snapshot. 

[1- Alias ECUB_007_P_003 cod.16829](https://github.com/icub-tech-iit/robot-snapshots/issues/30#issuecomment-1517554428)
[2- Alias ECUB_007_G_046 cod.16947 / Alias ECUB_007_G_042 cod.16949](https://github.com/icub-tech-iit/robot-snapshots/issues/30#issuecomment-1517733498)
[3- Alias ECUB_007_G_043 cod.16950 / Alias ECUB_007_G_045 cod.16948](https://github.com/icub-tech-iit/robot-snapshots/issues/30#issuecomment-1517822816)
[4- Alias ECUB_007_G_030 cod.16828](https://github.com/icub-tech-iit/robot-snapshots/issues/30#issuecomment-1523332007)

[image]

Ho anche tolto (soppressi) dal CAD dello snapshot i pezzi che non ci sono montati sul ergocub_sn_000 come ad es. il gruppo che blocca l'OLED e anche il supporto.

- ECUB_007_G_063 cod.17209
- ECUB_007_G_061 cod.17078

Elio75, 2023-05-02:
## Nota

Per il montaggio del gruppo maschera (maschera + magneti) Alias ECUB_007_G_062 cod.17208 sulla piattaforma ergocub_sn_000 bisognerà aggiornare la testa IC_014_A_100_ERGOCUB_HEAD cod.15742 rev 0 con i seguenti pezzi:

[image]

Alias IC_014_P_198 cod.16588 viene rimpiazzato con Alias IC_014_P_101 cod.17066
Alias IC_014_P_199 cod.16589 viene rimpiazzato con Alias IC_014_M_101 cod.17065
Alias ECUB_007_P_003 cod.16829 viene rimpiazzato con Alias  ECUB_007_P_047 cod.17062

Mick3Lozzo, 2023-05-02:
> ## Nota
> Per il montaggio del gruppo maschera (maschera + magneti) Alias ECUB_007_G_062 cod.17208 sulla piattaforma ergocub_sn_000 bisognerà aggiornare la testa IC_014_A_100_ERGOCUB_HEAD cod.15742 [...]

Yes, @Elio75 and also the covers `ECUB_007_G_001/002` will have a new revision and we will have to add the support `IC_014_G_118`, screw, magnets, and so on.

Elio75, 2023-05-02:
Per il resto del gruppo covers upper e lower body ho fatto un checkout in modo da verificare che non ci siano delle compenetrazione tra le covers e la meccanica della piattaforma sn_000, sembra che non ci siano dei problemi. 

CC @Fabrizio69 @Mick3Lozzo

closed 2023-07-31 by Elio75