icub-tech-iit/robot-snapshots #32: [ergocub_sn_000] [CAD update neck cable]

Elio75, 2023-05-08:
**Applicant:** *Elio Massa*, @Elio75 

**Snapshot name:** *ergocub_sn_000*

**Field of application:**
- [ ] Mechanics
- [ ] Electronics
- [ ] Wiring

## Request Update Description
Update Cad of the neck with Ø 0.81 uncoated steel cable. 

Material used for the construction of the cable

| Item / IITCODE | Quantity |  Description | Notes  |
|:---:|:---:|:---|:---|
|4722  | 4 | THIMBLE FOR CABLE n0.8mm |  |
| 4482  | 1 mtr | Stainless steel microcable, cable construction 7x7mm, nom. dia 0,81mm, not coated, AISI 304 |  |
|5964  | 1 | Head Cable B - 312 mm |  | 

## Definition of Done
CAD updated and the new part normalized to Wingst

## Info

[Costruzione tendini testa eCub s/n 000 prototipi Head cable "A" IIT CODE 5965MOD-Head cable "B" CODE 5964MOD](https://github.com/icub-tech-iit/proto/issues/2133#issue-1612944687)

Elio75, 2023-05-10:
The cables, cable head **A** RC_IIT_019_G_018 cod 5965 and cable head **B** RC_IIT_019_G_017 cod 5964 have been replaced with Alias cable IC_014_G_202 cod 17309 in the snapshot ergocub_SN_000

[image]

@Fabrizio69 @GiacomoVassallo13

Fabrizio69, 2023-05-10:
Hi @Elio75 , this tendon cod.17309 is an ASM but it's not a BOM in WinGST:

[image]

Fabrizio69, 2023-05-12:
@Elio75 have you checked the BOM?

[image]

Do we need to buy a new custom cable (IC_014_P_203)?????

Elio75, 2023-05-12:
The cable IC_014_G_202 cod 17309 is configured at the CAD level in the following form:

[image]

BOM 17309 at Wingst level:

[image]

CC @Fabrizio69

Elio75, 2023-05-24:
Ok ho risolto pare non abbia salvato il file che avevo modificato, ora è a posto 

[image]

CC @Fabrizio69

Fabrizio69, 2023-05-24:
Ciao @Elio75 puoi linkare a questa issue il commit fatto... grazie

closed 2023-06-07 by Elio75