icub-tech-iit/robot-snapshots #34: [ergoCub_SN_000] [Implementation for better cooling on the upper body]

Elio75, 2023-05-18:
**Applicant:** *Elio Massa*, @Elio75 

**Snapshot name:** *ergocub_sn_000*

**Field of application:**
- [ ] Mechanics
- [ ] Electronics
- [ ] Wiring

## Request Update Description
Implementation of three fans plus support on the torso of the robot so as to help with cooling and improve disipation.

[Optional table]

| Item / IITCODE | Quantity |  Description | Notes  |
|:---:|:---:|:---|:---|
| ECUB_004_P_054_SN_000 | 1 | Support fans |  |
| FAN_25X25X8 / 6784 | 3 | Fan 5VDC, screw mount, 25x25x8mm |  |
|  |  |  |  | 
|  |  |  |  |

## Definition of Done
CAD update design and assembly validation on ergocub_sn_000

## Info

[ergoCubSN000 – Motor temperature suspiciously high around some joints](https://github.com/robotology/icub-tech-support/issues/1528)

Fabrizio69, 2023-05-19:
Ciao @Elio75, in [your table](https://github.com/icub-tech-iit/robot-snapshots/issues/34#issue-1715833790), the description for cod.6784 - FAN_25X25X8 is wrong:

[image]

Here you can find the right description:

[image]

I corrected the table

fbiggi, 2023-05-19:
Eseguiti insieme a @maggia80 modifica sul particolare ECUB_007_P_010
N5 fori parte anteriore (vs petto) D 8mm
N2 fori D 8mm parte posteriore
N1 foro D10mm lato sx
N1 foro D10mm lato dx

[image]

@Fabrizio69 @pattacini

Elio75, 2023-05-19:
Ho canbiato la descrizione sia sullo snapshot che su CAD libraries

Elio75, 2023-05-22:
Abbiamo piazzato 3 ventole tra il busto (parte bassa) e il giunto Pitch del torso robot. La zona scelta si presentava meno critica rispetto ad altre come tipologia di lavoro e criticità.
Lo spazio all'interno molto fitto tra la cover i cavi e la rotazione del giunto YAW del torso. Per questo motivo bisognava configurare la posizione delle ventole e il disegno della piastra in modo tale da non avere compenetrazioni sia con le cover sia con la meccanica del giunto PITCH durante la rotazione dello YAW.

[image]

Fabrizio69, 2023-05-22:
Ciao @Elio75, riporterei sullo snapshot anche la modifica fatta da Filippo sulla cover in silicone (vedi [commento](https://github.com/icub-tech-iit/robot-snapshots/issues/34#issuecomment-1554486424))

Elio75, 2023-05-22:
> riporterei sullo snapshot anche la modifica fatta da Filippo sulla cover in silicone

Fatto

[image]

Fabrizio69, 2023-05-24:
Ciao @Elio75 , sarebbe utile riportare il riferimento del commit, il codice e l'alias generato. Questo farebbe capire subito cosa è stato fatto.

In questo caso specifico, ho scaricato lo snapshot del robot, sono andato a cercare il particolare e vedo che:

[image]

Hai creato una nuova revisione del codice ECUB_007_P_010 (rev.A1) contrariamente a quanto previsto dalla [snaphot procedure](https://github.com/icub-tech-iit/procedures/wiki/Robot-Snapshot#to-avoid-in-snapshot-repository).

Per scrupolo poi ho scaricato la repository del CAD e vedo che in realtà non hai cambiato la revisione.
Pensavo questa cosa perchè a WinGST esiste solo questo:

[image]

Riporto qui il CAD di ergoCub1.1:

[image]

Quindi questa cosa per favore va sistemata.

Altra cosa che noto che non va bene è che questo componente è un PLAS, ma questo vedo di segnalarlo a chi sta gestendo le cover. Grazie mille

Elio75, 2023-05-24:
Creato nuovo alias [ECUB_007_P_010_SN_000](https://github.com/icub-tech-iit/robot-snapshots/issues/34#issuecomment-1557106627) al capuccio in gomma del collo. La parte è stata modificata solo sullo snapshot di ergocub_sn_000 e di conseguenza il suo alias come indica la procedura.

CC @Fabrizio69

Fabrizio69, 2023-06-12:
Ciao @Elio75 direi che a questo punto si possa chiudere questa issue... grazie mille

closed 2023-06-12 by Elio75