icub-tech-iit/robot-snapshots #36: [ergoCub_sn_001] Creazione dello snapshot per ergoCub_sn_001

Elio75, 2023-07-31:
**Applicant:** *Elio Massa*, @Elio75 

**Snapshot name:** *ergocub_sn_001*

**Field of application:**
- [ ] Mechanics
- [ ] Electronics
- [ ] Wiring

## Request Update Description
Creare lo snapshot della piattaforma ergocub_sn_001

[Optional table]

| Item / IITCODE | Quantity |  Description | Notes  |
|:---:|:---:|:---|:---|
| ECUB/16864.A1 | 1 | ergoCub 1.1 , assembly |  |
|  |  |  |  |
|  |  |  |  | 
|  |  |  |  |

## Definition of Done
Fare backup di tutto il robot dentro la repo snapshot &#x2192; branch ergocub_sn_001

pattacini, 2023-07-31:
cc @Mick3Lozzo

Elio75, 2023-08-01:
Weight 

[image]

Fabrizio69, 2023-08-01:
Ciao @Elio75 con lo stesso CAD che userai per creare lo snapshot di eCub1.1 S/N:001 (oggetto di questa issue), dovrai creare anche lo snapshot per eCub1.1 S/N:002

fiorisi, 2023-08-01:
@Elio75 allegherei come riferimento anche i commit di cad-mechanics e cad-libraries relativi allo snapshot.

Elio75, 2023-08-01:
@fiorisi ti riferisci all'ultimo commit fatto?

fiorisi, 2023-08-01:
> > allegherei come riferimento anche i commit di cad-mechanics e cad-libraries relativi allo snapshot.
> 
> @fiorisi ti riferisci all'ultimo commit fatto?

Esatto l'ultimo commit a cui erano state aggiornate le repository quando hai fatto lo snapshot.

Elio75, 2023-08-01:
## Ultimo commit di riferimento

- CAD-mechanics
  &#x21B3; [correcting cover colors for painting documentation](https://github.com/icub-tech-iit/cad-mechanics/commit/d1ee9768b963a0bb95114f278bf9b2382213d495)

- CAD-libraries
  &#x21B3; [updated wrong weight](https://github.com/icub-tech-iit/cad-libraries/commit/be5d51953086f5d8076a4125d5d8eab4949a5a34)

CC @fiorisi

Elio75, 2023-08-01:
Fixed revision ic_028_g_030 from A1 to A2

@fiorisi

maggia80, 2023-08-01:
Logic Schematics:  imported in wingst <html>
<body>

17256 | 17256 | 1.1.0 |   | Custom kit | ergoCub1.1 electronic parts - full robot
-- | -- | -- | -- | -- | --

</body>
</html> 

PDF: [ergoCub1.1_Logic_17256_1.1.0.pdf](https://github.com/icub-tech-iit/electronics-wiring/blob/master/ergocub1/ergocub1.1/pdf/ergoCub1.1_Logic_17256_1.1.0.pdf) commit: https://github.com/icub-tech-iit/electronics-wiring/commit/805300cb2bb05d5b18f7b9bf5b0b14f4a5adbf3e#diff-280bcf8c74b662b42cc3cc7d9086234178d8b0145a7701f85f8dc824f478fef2

Fabrizio69, 2023-08-29:
Ciao @maggia80, I see now that in the logic schematic BOM (electronic parts), there are obsolete cod.16509 and cod.16508 replaced by the same component cod.17398. Is this correct?

Elio75, 2024-01-11:
@Fabrizio69 la chiudo