icub-tech-iit/robot-snapshots #37: ergoCub1.1_sn_002 Snapshot creation

Fabrizio69, 2023-08-01:
**Applicant:** *Fabrizio Larosa*, @Fabrizio69 

**Snapshot name:** *ergocub1.1_sn_002*

**Field of application:**
- [X] Mechanics
- [ ] Electronics
- [ ] Wiring

## Request Update Description
Create the snapshot for eCub1.1 S/N:002 platform

| IITCODE/ALIAS | Quantity |  Description | Notes  |
|:---:|:---:|:---|:---|
| 16864.A1/ECUB | 1 | ergoCub1.1 assembly |  |

## Definition of Done
The snapshot is created (new branch in the snapshot repo) and the BOM is imported in WinGST

Elio75, 2023-08-01:
@Fabrizio69 BOM creata di ecub1.1 cod. 16864.A1

Elio75, 2023-08-01:
Creato snapshot ergocub1.1_SN_002
Commit linkato al commento come si può vedere sotto ↓

Elio75, 2023-08-01:
## Last reference commit

- CAD-mechanics
  &#x21B3; [Fix typo in revision (from A1 to A2).](https://github.com/icub-tech-iit/cad-mechanics/commit/0913f5666d8d972a9e1107650639f4d6da9198b3)

- CAD-libraries
  &#x21B3; [updated wrong weight](https://github.com/icub-tech-iit/cad-libraries/commit/be5d51953086f5d8076a4125d5d8eab4949a5a34)

CC @fiorisi

Fabrizio69, 2023-08-04:
I reopen this issue for the electronic and wiring snapshots. I add also all reference info in the OP...

Fabrizio69, 2023-08-04:
Ciao @maggia80 , in the electronics BOM (logic schematics) there is AMO1 rev.A (cod.6778.A), but this board now is not available (AMO1 is on the eCub1.1 S/N:001 platform). We have AMO1 rev.B:

[image]
 
Does this board replacement my affect firmware/software or something else?

cc: @Nicogene @pattacini

pattacini, 2023-08-04:
>Does this board replacement my affect firmware/software or something else?

Probably not, but worth checking.
Maybe @davidetome knows it.

davidetome, 2023-08-04:
> Probably not, but worth checking.
Maybe @davidetome knows it.

I think so but @maggia80 can surely confirm,

cc @Fabrizio69

maggia80, 2023-08-08:
AMO1 rev.B is fine as well. The EEPROM is used by the iC-MU sensor for storing data about the calibration and some other options. It does not affect the communication between the AMO1 and the EMS4, so there isn't any firmware/software change needed.

Fabrizio69, 2023-08-29:
Thanks @maggia80! If there isn't any firmware/software change needed , we can mix on this robot cod.6778 rev.A (latest available and already withdrawn automatically, from the warehouse) with cod.6778 rev.B . OK we'll proceed in this way. Thanks

Fabrizio69, 2023-08-30:
According to [this comment](https://github.com/icub-tech-iit/purchases/issues/2376#issuecomment-1699438503): the MMA boards (cod.16872.A) are replaced by cod.16872.B.

Elio75, 2023-09-01:
readme electronic ergocub_sn_002 (MMA / AMO1) updated 

cc @Fabrizio69

closed 2023-12-09 by Fabrizio69