icub-tech-iit/robot-snapshots #40: [ergoCub_sn_001/002] [waist front and back covers and supports]

Mick3Lozzo, 2023-08-25:
**Applicant:** *Michele Gesino*, @Mick3Lozzo 

**Snapshot name:** *ergoCub_sn_001*, *ergoCub_sn_002*

**Field of application:**
- [ ] Mechanics
- [ ] Electronics
- [ ] Wiring

## Request Update Description
As per [this comment](https://github.com/icub-tech-iit/project-ergocub/issues/121#issuecomment-1664020830) at last we decided to:

- Modify the front waist and IMU cover.
- Modify the IMU support.
- Remove from the CAD the back waist cover (that has never been sent to production, though). 

[Optional table]

| Item / IITCODE | Quantity |  Description | Notes  |
|:---:|:---:|:---|:---|
|  |  |  |  |
|  |  |  |  |
|  |  |  |  | 
|  |  |  |  |

## Definition of Done
[insert here the definition of done for this task]

Mick3Lozzo, 2023-08-25:
@Fabrizio69 before working on this issue we should send the components to production. 
I have already sent them to production n° 3 `17620 - ECUB_003_G_155`, and I will put them in the warehouse when they are available.

Elio75, 2023-08-29:
@Mick3Lozzo sui sn_001 / sn_002: 

-  >Sostituisci 17297 - ECUB_003_G_152con 17620 - ECUB_003_G_155in 15789.A2 - ECUB_003_G_020.
   - Vorresti dire sostituisci ECUB_003_G_153 con ECUB_003_G_155 ? vero?

@pattacini @Fabrizio69

pattacini, 2023-08-29:
/remind September 11 @Mick3Lozzo please address @Elio75 [request](https://github.com/icub-tech-iit/robot-snapshots/issues/40#issuecomment-1697340962).

Elio75, 2023-08-29:
Modifiche CAD sn_002 &#x2192; **FATTO**

@Mick3Lozzo @Fabrizio69

Elio75, 2023-08-29:
Modifiche CAD sn_001 → FATTO

@Mick3Lozzo @Fabrizio69

Elio75, 2023-08-31:
@Mick3Lozzo 

 >Update production BOMs and orders

see [comment](https://github.com/icub-tech-iit/robot-snapshots/issues/40#issuecomment-1693052770)

Elio75, 2023-09-14:
Chiudo

closed 2023-09-14 by Elio75