icub-tech-iit/robot-snapshots #41: ergoCub_sn_001/002 - AEA3 support and neck pitch hard stop

Mick3Lozzo, 2023-08-25:
**Applicant:** *Michele Gesino*, @Mick3Lozzo 

**Snapshot name:** *ergoCub_sn_001*, *ergoCub_sn_002*

**Field of application:**
- [ ] Mechanics
- [ ] Electronics
- [ ] Wiring

## Request Update Description
As per [this issue](https://github.com/icub-tech-iit/fix/issues/1426) to the neck AEA3 group, the same modification has to be applied on the ergoCub snapshots too.

[image]

 > **NOTE**:
 > This modification is still a work in progress, has to be tested and eventually sent to production.

[Optional table]

| Item / IITCODE | Quantity |  Description | Notes  |
|:---:|:---:|:---|:---|
|  IC_014_P_205 / 17636 | 1 |  AEA3 holder and neck pitch hard stop |  |
| IC_014_G_182 - 15718.A3 /  |  1 | neck tilt AA encoder PCB support |  |

## Definition of Done

- [ ] In `IC_014_G_182 - 15718.A3` remove `15737 - IC_014_P_181`, `17194 - RC_IIT_010_P_092` and the screw `2440 - V2-6--_-_ISO7046-1_CH`.

Mick3Lozzo, 2023-08-25:
@Elio75 @Fabrizio69 this solution still has to be tested and the components have not ever been sent to production.

Nicogene, 2023-08-30:
Putting on hold until @Mick3Lozzo returns

/remind 11 september

Elio75, 2023-09-05:
Comunque prima di modificare / unificare dei pezzi in modo da semplificare il sistema , io proverei a migliorare solo il supporto

Elio75, 2023-10-09:
## CAD update of the neck ergocub_sn_002

closed 2023-10-13 by Elio75