icub-tech-iit/robot-snapshots #42: ergoCub_sn_001/002 - neck roll motor holder

Mick3Lozzo, 2023-08-25:
**Applicant:** *Michele Gesino*, @Mick3Lozzo 

**Snapshot name:** *ergoCub_sn_001*, *ergoCub_sn_002*

**Field of application:**
- [ ] Mechanics
- [ ] Electronics
- [ ] Wiring

## Request Update Description

[image]

Modify 2607.A1 - RC_IIT_010_P_023 as per [this comment](https://github.com/icub-tech-iit/proto/issues/2456#issuecomment-1690123817) and [this other comment](https://github.com/icub-tech-iit/proto/issues/2456#issuecomment-1690087852).

[Optional table]

| Item / IITCODE | Quantity |  Description | Notes  |
|:---:|:---:|:---|:---|
| RC_IIT_010_P_023 / 2607.A1 | V2 neck pan motor holder |  |  |
|  |  |  |  |

## Definition of Done

- [ ] Define with @GiacomoVassallo13 the modifications to `2607.A1`.
- [ ] Update official CAD, increase `2607` revision to `A2`.
- [ ] Update snapshots `001` and `002` @Elio75 
- [ ] Send components to production.
- [ ] Update BOMs and orders @Fabrizio69.

Mick3Lozzo, 2023-08-25:
On hold, modifications still to be defined, see [this comment](https://github.com/icub-tech-iit/proto/issues/2456#issuecomment-1693193110).

Elio75, 2023-08-28:
Scusate ma forse bastava girare il pezzo di 180° per far passare il flat sul davanti senza dover fare nessuna modifica.

[image]

[image]

@Mick3Lozzo

Elio75, 2023-08-31:
Ho riportato la modifica su ergocub_sn_001

Elio75, 2023-08-31:
[](url)

Elio75, 2023-08-31:
Insieme a @fiorisi abbiamo deciso di: 

- fare una prova a CAD girando il pezzo RC_IIT_010_P_023  insieme al motore di 180° in modo da verificare che no ci siano problemi di compenetrazione o di montaggio degli altri gruppi permettendo ai cavi di uscire sul davanti. Questo potrebbe risolvere il problema visto che il pezzo si presenta simetrico in quasi la sua totaslità e cosi evitare altre modifiche su questo ultimo.

CC @Fabrizio69 @GiacomoVassallo13 @Nicogene @Gandoo

Elio75, 2023-09-01:
## Prova CAD

- Ho sovraposto l'assieme IC_014_G_181 uno a 0° e l'atro a 180°, come si può vedere dalla immagiene i fori di fissaggio e fori spinati combaciano entrambi lati.

[image]

- La parte inferiore del collo rispetto a quella superiore viene montata mediante due spine controposte e il anello clamp che li tiene insieme, perciò la parte inferiore può essere girata senza problemi visto che l'accopiamento viene fatto tra la wave generator e la flexpline del HD e le due superficie di contatto di entrambi gruppi.

[image]

Mick3Lozzo, 2023-09-11:
> ## Prova CAD
> * Ho sovraposto l'assieme IC_014_G_181 uno a 0° e l'atro a 180°, come si può vedere dalla immagiene i fori di fissaggio e fori spinati combaciano entrambi lati.
> 

@Elio75 ripristino la versione precedente di `RC_IIT_010_P_023` incluso il disegno, e monto ribaltato di 180° `RC_IIT_010_G_011` dentro l'assieme `IC_014_G_181`.

Mick3Lozzo, 2023-09-11:
@Elio75 fatto, dovrebbe essere tutto a posto.

Elio75, 2023-09-11:
CAD a posto collo girato... ora farò lo stesso sugli snapshot

Fabrizio69, 2023-09-11:
Ciao @Elio75, ma sullo snapshot di eCub1.1 S/N:001 questa modifica deve essere riportata?
Il collo dell'iCub1.1 S/N:002 deve ancora essere assemblato, ma quello del S/N:001 è già costruito...

Elio75, 2023-09-11:
Si bisognerà farlo sul sn_001 / 002

Elio75, 2023-09-12:
Il supporto motore del collo IC_014_G_181 è stato girato con il cavo uscente sul davanti

Elio75, 2023-09-12:
## Snapshot ergocub_sn_001

Ho ripristinato a CAD il collo del sn_001 rispetto quello fisico già montato e colaudato. Il pezzo RC_IIT_010_P_023 cod2607.A1 non è stato girato rispetto al suo omologo sn_002 ma il motore (YAW) si portando sul davanti l'uscita dei cavi.

Elio75, 2023-09-27:
Il supporto motore neck roll girato di 180° è stato montato da @GiacomoVassallo13 sull'assieme testa ergocub_sn_002. Il montaggio è avvenuto con successo, non sono state  risscontrate anomalie.

https://github.com/icub-tech-iit/robot-snapshots/assets/53298662/e438a784-e7e7-4221-be6c-73d3aaf8f46f

Elio75, 2023-09-29:
Direi che la modifica di montaggio va bene, la issue si può chiudere

closed 2023-10-09 by Elio75