icub-tech-iit/robot-snapshots #45: [ergocub_sn_002] Calf support inspection holes

Elio75, 2023-09-25:
**Applicant:** *Elio Massa*, @Elio75 

**Snapshot name:** *ergocub_sn_002*

**Field of application:**
- [ ] Mechanics
- [ ] Electronics
- [ ] Wiring

## Request Update Description
Add 4 inspection holes on the calf support of the legs (Alias ECUB_003_P_134 cod16111) to reach the fixing screws of the mechanical stop and harmonic driver.

[image]

[Optional table]

| Item / IITCODE | Quantity |  Description | Notes  |
|:---:|:---:|:---|:---|
| Alias ECUB_003_P_134 cod16111 | 1 | Support for calf |  |
|  |  |  |  |
|  |  |  |  | 
|  |  |  |  |

## Definition of Done
Update CAD ergocub_sn_002

Elio75, 2023-09-26:
Ho aggiornato il CAD con delle viti ad alta resistenza 10.9 / 12.9

CC @Fabrizio69 @fbiggi

Elio75, 2023-09-26:
Chiudo

Fabrizio69, 2023-09-26:
Ciao @Elio75 avrei indicato anche il codice (oppure i codici) delle viti che hai sostituito

Elio75, 2023-09-26:
[Fatto](https://github.com/icub-tech-iit/robot-snapshots/issues/45#issue-1910914665)

Fabrizio69, 2023-09-27:
Grazie @Elio75 a questo punto direi si possa chiudere questa issue.

closed 2023-09-27 by Fabrizio69