icub-tech-iit/robot-snapshots #48: ergoCub1.1_sn_001 Added CAN_TERM boards (cod.17662.A)

Fabrizio69, 2023-10-10:
**Applicant:** *Fabrizio Larosa*, @Fabrizio69 

**Snapshot name:** *ergoCub1.1_sn_001*

**Field of application:**
- [ ] Mechanics
- [X] Electronics
- [X] Wiring

## Request Update Description
According to [these changes](https://github.com/icub-tech-iit/proto/issues/2542) for eCub1.1 S/N:001 the harness and the logic diagrams have to change

[Optional table]

| Item / IITCODE | Quantity |  Description | Notes  |
|:---:|:---:|:---|:---|
|  |  |  |  |
|  |  |  |  |
|  |  |  |  | 
|  |  |  |  |

## Definition of Done
The snapshot is updated (New harness and new Logic diagrams)

AntonioAzocar, 2023-10-30:
Added the Can_term board (cod. 17622.A) between EB21 and RFE (**HEAD**)

[image]

cc @Fabrizio69 @maggia80

AntonioAzocar, 2023-10-30:
Check here for the updated logic & harness diagrams:

- https://github.com/icub-tech-iit/electronics-wiring/commit/e87cc780e63e43fef0e5846e34aa3593712a770c

AntonioAzocar, 2023-10-30:
So, now I think we can close this issue @Fabrizio69

Fabrizio69, 2023-11-02:
Hi @AntonioAzocar, this issue is for eCub1.1 S/N:001 changes and I don't see changes in the snapshot for this robot... 

[image]

[image]

AntonioAzocar, 2023-11-03:
Done!
Now you can see the logic & harness diagrams updated

- https://github.com/icub-tech-iit/robot-snapshots/commit/cf686108cff6f091a72b25e35621b9ba00eee82d

closed 2023-11-07 by AntonioAzocar