icub-tech-iit/robot-snapshots #49: [platform_sn_001] Supporto OLED  alias IC_014_G_118 cod.17307

Elio75, 2023-10-31:
**Applicant:** *Elio Massa*, @Elio75 

**Snapshot name:** *platform_sn_001*

**Field of application:**
- [ ] Mechanics
- [ ] Electronics
- [ ] Wiring

## Request Update Description
Sostituire sul sn_001 l'assieme ECUB_007_G_063 cod.17209 rev0 con la rev A1, all'interno ci sarà il support OLED della testa alias IC_014_G_118 cod.17307 in rev A7  

[Optional table]

| Item / IITCODE | Quantity |  Description | Notes  |
|:---:|:---:|:---|:---|
|IC_014_G_118 / cod.17307  | 1 | ergoCub, head covers, OLED locker |  |
|ECUB_007_G_063 / cod.17209  | 1 | ergoCub, head covers, OLED locker assembly |  |
|  |  |  |  | 
|  |  |  |  |

## Definition of Done
CAD aggiornato

Elio75, 2023-10-31:
**CAD aggiornato**

è stato sostituito sullo snapshot l'assieme ECUB_007_G_063 / cod.17209 con la rev A1, all'interno c'è anche la nuova revisione in A7 del supporto OLED alias IC_014_G_118 / cod.17307.

**NOTA:** sul robot è stata montato il supporto IC_014_G_118 / cod.17307. rev A6 ma sul CAD dello snapshot rimane la rev A7.

CC @Fabrizio69 @fbiggi @Mick3Lozzo

Fabrizio69, 2023-10-31:
Ciao @Elio75, se sul robot eCub1.1 S/N:001 abbiamo montato il cod.17307.A6 (IC_014_G_118 rev. A6) sul relativo snapshot andrai a montare la Rev.A6 non la Rev.A7. La rev.A6 non è più disponibile sul CAD, ma la puoi trovare [qui](https://github.com/icub-tech-iit/cad-mechanics/commit/c843c0ce1fccc13576bfaee6a9b51a2273cf32c8) dove @Mick3Lozzo la rilasciava (vedi anche [questo commento](https://github.com/icub-tech-iit/proto/issues/2492#issuecomment-1735674058))

Elio75, 2023-10-31:
Ho aggiornato il CAD dello snapshot eCub1.1 S/N:001. L'alias  IC_014_G_118 / cod.17307 è stato modificato con la rev A6 invece della rev A7. [vedi commento](https://github.com/icub-tech-iit/robot-snapshots/issues/49#issuecomment-1787504985).

closed 2023-11-06 by Elio75