icub-tech-iit/robot-snapshots #56: [ergocub_sn_002] New support AEA -Fore arm - Alias IC_021_G_186 cod. 17682

Elio75, 2023-11-27:
**Applicant:** *Elio Massa*, @Elio75 

**Snapshot name:** *ergocub_sn_002*

**Field of application:**
- [ ] Mechanics
- [ ] Electronics
- [ ] Wiring

## Request Update Description
Replace AEA holder aka IC_021_G_049 cod.16366 with alias IC_021_G_186 cod. 17682

[Optional table]

| Item / IITCODE | Quantity |  Description | Notes  |
|:---:|:---:|:---|:---|
| 17682 /IC_021_G_186 | 2 | lower arm, frame cover support with inserts |  |
|  |  |  |  |
|  |  |  |  | 
|  |  |  |  |

## Definition of Done
snapshots CAD update

Elio75, 2023-11-28:
Update CAD snapshot ergocub_sn_002

Fabrizio69, 2023-11-28:
See the [changes on the robot](https://github.com/icub-tech-iit/proto/issues/2589#issuecomment-1824500433)

closed 2023-12-01 by Elio75