icub-tech-iit/robot-snapshots #57: [ergocub_sn_000 - ] [Update feet soles]

fiorisi, 2023-11-28:
**Applicant:** *Masa Salvi*, @salvi-mattia 

**Snapshot name:** 
- *ergocub_sn_000*
- *ergocub_sn_001* -> `Note`: assemble also on the real robot
- *ergocub_sn_002* -> `Note`: assemble also on the real robot

**Field of application:**
- [x] Mechanics
- [ ] Electronics
- [ ] Wiring

## Request Update Description
Replace the feet soles as described in https://github.com/icub-tech-iit/project-ergocub/issues/132#issuecomment-1829875090. 

[Optional table]

| Alias | Quantity for one robot |  Description | Rev |
|:---:|:---:|:---|:---|
| ECUB_003_P_047 | 4 |  | A1 |
| ECUB_003_P_050 | 4 |  | A1 |
|  |  |  |  | 
|  |  |  |  |

[image]

## Definition of Done
- [ ] snapshots SN000, SN001 and SN002 updated
- [ ] Real robot SN001 and SN002 updated

Elio75, 2023-11-30:
**Update CAD snapshot ergocub_sn_001**

Elio75, 2023-11-30:
**Update CAD snapshot ergocub_sn_002**

fiorisi, 2023-12-01:
>  Real robot SN001 and SN002 updated

@Elio75 concerning this point did you open a dedicated issue?

Elio75, 2023-12-01:
Scusa l'ho chiusa per sbaglio

Fabrizio69, 2023-12-09:
Gli snapshot sono stati aggiornati, procediamo con la chiusura di questa issue rimandando a [ergoCub1.1 S/N:001 - Install the new soles (cod.16018.A1, cod.16587.A1)](https://github.com/icub-tech-iit/proto/issues/2650) per l'installazione sul robot

closed 2023-12-09 by Fabrizio69

Elio75, 2023-12-13:
Ho aggiunto una NOTA (TOOL) sul albero modello per l'uso del tool IT_004_G_001 durante il montaggio delle suole dei piedi. Questo sui robot ergocub_sn_001 / ergocub_sn_002

[image]

fiorisi, 2023-12-13:
> Ho aggiunto una NOTA (TOOL) sul albero modello per l'uso del tool IT_004_G_001 durante il montaggio delle suole dei piedi. Questo sui robot ergocub_sn_001 / ergocub_sn_002

Ottimo @Elio75. Per favore indica in modo esplicito l'alias del tool che deve essere usato. 
See the dedicated note in the [Annotation Features](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#annotation-feature-creation-procedure) section:

[image]

Elio75, 2023-12-13:
ergoCub_sn_001 commit
- [modified NOTE](https://github.com/icub-tech-iit/robot-snapshots/commit/b7cb17f25cf8981b4522ef3a179a794a6d3b1de9)
- [added note for Tool](https://github.com/icub-tech-iit/robot-snapshots/commit/dd3c063de8f6d5154d08770a43f3314b40b0b97e)