icub-tech-iit/robot-snapshots #59: [ergoCub_sn_001/002] [Modify ergoCub neck rubber cap]

Mick3Lozzo, 2024-01-08:
**Applicant:** *Michele Gesino*, @Mick3Lozzo 

**Snapshot name:** *ergoCub_sn_001*, *ergoCub_sn_002*

**Field of application:**
- [ ] Mechanics
- [ ] Electronics
- [ ] Wiring

## Request Update Description
As per [this issue](https://github.com/icub-tech-iit/project-ergocub/issues/133) following this [proto feedback](https://github.com/icub-tech-iit/proto/issues/2410), I updated to `A1` the neck rubber cap to avoid interferences.

[image]

[image]

### Annotations

- The item has already been ordered in Rev. 0, and it will be manually modified by @fbiggi [as previously done](https://github.com/icub-tech-iit/proto/issues/2410#issuecomment-1832046844) to make it equivalent to the rev. A1.

- I didn't do this reworking 

[image]

but I enlarged a little the neck passage.

| Item / IITCODE | Quantity |  Description | Notes  |
|:---:|:---:|:---|:---|
| 16899 | 1 | ergoCub covers, neck, rubber cap | see below |

## Definition of Done

- `16899 - ECUB_007_P_010` replaced with `16899.A1 - ECUB_007_P_010` in the assembly `16894.A1 - ECUB_007_G_054`.
- Update orders @Fabrizio69 
- Modify the already ordered rev. 0 components @fbiggi

Mick3Lozzo, 2024-01-08:
@Fabrizio69 I requested the update for `ergoCub_sn_001` and `ergoCub_sn_002`, but maybe we have to include even the future 003/004?

maggia80, 2024-01-08:
@Mick3Lozzo I'll add the part in the two snapshots ergoCub_sn_001 and ergoCub_sn_002 and we could do the rework of this part if we have it in the warehouse. Since it's in the master MCAD already, other snapshots of the robot will include this modification.

Fabrizio69, 2024-01-08:
Ciao @Mick3Lozzo , ciao @Elio75,
the new component cod.16899.A1 ECUB_007_P_010 rev.A1 it's the same created in the ergoCub1.0 S/N:000 snapshot (see the [link](https://github.com/icub-tech-iit/robot-snapshots/issues/34#issuecomment-1556861773))? 
Could it be useful to unify the components: ECUB_007_P_010 rev.A1 and ECUB_007_P_010_SN_000?

Mick3Lozzo, 2024-01-09:
> Ciao @Mick3Lozzo , ciao @Elio75, the new component cod.16899.A1 ECUB_007_P_010 rev.A1 it's the same created in the ergoCub1.0 S/N:000 snapshot (see the [...]

@Fabrizio69 I confirm, it's the same component. The can be unified.

cc @Elio75

Elio75, 2024-01-11:
- [ ] Update CAD ergoCub_sn_001 with ECUB_007_P_010 cod.16899 rev A2

Elio75, 2024-01-11:
- [ ] Aggiornamento CAD ergoCub_sn_002 con ECUB_007_P_010 cod.16899 rev A2

Fabrizio69, 2024-01-15:
Hi @Elio75, hi @Mick3Lozzo, with [this change](https://github.com/icub-tech-iit/robot-snapshots/issues/59?notification_referrer_id=NT_kwDOALeVn7M5MDA1MzU3MzQ5OjEyMDMxMzkx&notifications_query=repo%3Aicub-tech-iit%2Frobot-snapshots#issuecomment-1887188286) we have to update all parts available in the warehouse (see [this issue](https://github.com/icub-tech-iit/proto/issues/2698))

closed 2024-01-15 by Fabrizio69