icub-tech-iit/robot-snapshots #63: [ecub_sn_002] - Update hands

Elio75, 2024-01-15:
**Applicant:** *Elio Massa*, @Elio75 

**Snapshot name:** *ecub_sn_002*

**Field of application:**
- [ ] Mechanics
- [ ] Electronics
- [ ] Wiring

## Request Update Description
Update hands Left / Right:
- Ring and little finger mechanism
- Tendons registers
- Thumb abduction
- Proximal CNC part

[Optional table]

| Item / IITCODE | Quantity |  Description | Notes  |
|:---:|:---:|:---|:---|
|  |  |  |  |
|  |  |  |  |
|  |  |  |  | 
|  |  |  |  |

## Definition of Done
Update CAD ergocub_sn_002

Elio75, 2024-01-15:
- **Thumb abduction**

| Item / IITCODE | Quantity |  Description | Notes  |
|:---:|:---:|:---|:---|
|IC_028_P_144 / 17724 | 2 | hands, metacarpus lever 2 |  |
|IC_028_P_146 / 17725 | 2 | Hand, thumb, shaft with milling plane for grub |  |
|IC_028_P_145 / 17726 | 1 | hands, left metacarpus |  | 
|IC_028_M_145 / 17728 | 1 | hands, right metacarpus |  |
|IC_028_P_143 /17727  | 8 | hands, Adjustment screws for tendons |  |

W.P.

Elio75, 2024-01-15:
- Ring and little finger mechanism

| Item / IITCODE | Quantity |  Description | Notes  |
|:---:|:---:|:---|:---|
|IC_028_M_038 |  |  |  |
|IC_028_M_039 |  |  |  |
|IC_028_P_034 |  |  |  | 
|IC_028_P_035 |  |  |  |
|IC_028_P_036 |  |  |  |
|IC_028_P_067 |  |  |  |
|IC_028_P_070 |  |  |  |
|IC_028_P_071 |  |  |  |
|IC_028_P_073 |  |  |  |
|IC_028_P_082 |  |  |  |
|IC_028_P_083 |  |  |  |
| |  |  |  |

Elio75, 2024-01-15:
BOM mano sinistra e mano destra (CAD-Mechanics)
[ic_028_a_001_cad-mech_sx.txt.xlsx](https://github.com/icub-tech-iit/robot-snapshots/files/13941293/ic_028_a_001_cad-mech_sx.txt.xlsx)
[ic_028_a_002_cad-mech_dx.txt.1.xlsx](https://github.com/icub-tech-iit/robot-snapshots/files/13941295/ic_028_a_002_cad-mech_dx.txt.1.xlsx)

Elio75, 2024-01-22:
@Fabrizio69 direi che si può chiudere.

Fabrizio69, 2024-01-23:
Ciao @Elio75,
nel file [ic_028_a_001_cad-mech_sx.txt.xlsx](https://github.com/icub-tech-iit/robot-snapshots/files/13947725/ic_028_a_001_cad-mech_sx.txt.xlsx) hai riportato in verde le `migliorie`, quindi immagino siano i codici nuovi che hai aggiunto. Tra questi hai indicato il codice IC_028_P_070 che però non risulta nè in `Thumb abduction`, nè in `Tendon registers` e nè in `Ring and little finger mechanism`.

Sempre nello stesso file hai evidenziato in giallo i codici WinGST, tranne uno. Cosa significa? Non ci sono indicazioni.

La stessa osservazione anche per il file [ic_028_a_002_cad-mech_dx.txt.1.xlsx](https://github.com/icub-tech-iit/robot-snapshots/files/13947727/ic_028_a_002_cad-mech_dx.txt.1.xlsx).

Verifica un attimo e poi procediamo con la chiusura della issue. Grazie mille

Elio75, 2024-01-23:
Ok ho sistemato i file excel.