icub-tech-iit/robot-snapshots #64: [ergoCub_sn_001 / ergoCub_sn_001] [head frame general assembly - new board Orin NX]

Mick3Lozzo, 2024-01-15:
**Applicant:** *Michele Gesino*, @Mick3Lozzo 

**Snapshot name:** *ergoCub_sn_001*, *ergoCub_sn_002*

**Field of application:**
- [ ] Mechanics
- [ ] Electronics
- [ ] Wiring

## Request Update Description
Update the head with the new Orin NX board and all necessary supports, see [this issue](https://github.com/icub-tech-iit/ergocub-design-head/issues/121) and [this other issue](https://github.com/icub-tech-iit/ergocub-design-head/issues/120).

[Optional table]

| Item / IITCODE | Quantity |  Description | Notes  |
|:---:|:---:|:---|:---|
|  |  |  |  |
|  |  |  |  |
|  |  |  |  | 
|  |  |  |  |

## Definition of Done
[insert here the definition of done for this task]

### WIP

Mick3Lozzo, 2024-01-15:
@Elio75 ho iniziato a scrivere questa issue intanto che aspetto di effettuare i test per portarmi avanti, poi cancellerò anche questo commento.

[Ho aperto un ticket](https://github.com/icub-tech-iit/tickets/issues/3301).

**Per adesso** non c'è da fare assolutamente nulla.

@Fabrizio69 @Nicogene

Mick3Lozzo, 2024-01-17:
@Elio75 as per [this comment](https://github.com/icub-tech-iit/ergocub-design-head/issues/121#issuecomment-1895924404) we can proceed.

@Nicogene @Fabrizio69

Mick3Lozzo, 2024-01-18:
> Ciao @Mick3Lozzo , why do you want to change the ergoCub1.1 S/N:001 snapshot (`ergoCub_sn_001`)?
> 
> cc: @Elio75 @Nicogene

@ I understood that we would also update the head electronics of the existing robot, if this is not our intention then let's change the title of this issue.

cc @Elio75

Elio75, 2024-01-18:
Inizierò con il sn_002 poi se bisognerà fare il sn_001 lo farò.

@Nicogene @Fabrizio69

Fabrizio69, 2024-01-18:
> @ I understood that we would also update the head electronics of the existing robot, if this is not our intention then let's change the title of this issue.
> 
> cc @Elio75

@Mick3Lozzo, at this time, as defined by the Board (January 9, 2024), these changes will only be made on ergoCub1.1 S/N:002. We can change the issue title...

cc: @Elio75

Elio75, 2024-01-18:
Il CAD di ergocub_sn_002 è stata aggiornata secondo la [lista](https://github.com/icub-tech-iit/robot-snapshots/issues/64#issue-2081769140) di @Mick3Lozzo.

[image]

Elio75, 2024-01-22:
@Fabrizio69  direi che si può chiudere.