icub-tech-iit/robot-snapshots #68: eCub_sn_002 - Add ferrites to TDB cables

Fabrizio69, 2024-02-09:
**Applicant:** *Fabrizio Larosa*, @Fabrizio69 

**Snapshot name:** *eCub_sn_002*

**Field of application:**
- [ ] Mechanics
- [X] Electronics
- [X] Wiring

## Request Update Description
Change the diagrams for eCub1.1 S/N:002, according to [this issue](https://github.com/icub-tech-iit/proto/issues/2738#issue-2125289177), where the ferrites cod.17820 are added to the TDB cables.

[Optional table]

| Item / IITCODE | Quantity |  Description | Notes  |
|:---:|:---:|:---|:---|
|  |  |  |  |
|  |  |  |  |
|  |  |  |  | 
|  |  |  |  |

## Definition of Done

- [ ] New diagrams are in the snapshot repository: [add here the link]

AntonioAzocar, 2024-02-12:
Hi! I updated the schematics of `ergoCub1.1 S/N:002` with the addition of ferrites (code **17820**) to the **TDB** cables, the replacement of the **Xavier** board with the **Orin** board and also fixed other minor things (heat shrink, cable lengths...)

Here, the commit with the updated schematics:

- https://github.com/icub-tech-iit/robot-snapshots/commit/636d7b6898fa8bd60be1b69e7e6f7f8c1e91ef46

cc @Fabrizio69

Fabrizio69, 2024-02-13:
We can close thi issue. Thank you @AntonioAzocar

closed 2024-02-13 by Fabrizio69